mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #482: Forced Charge suddenly not working for no apparent reason

math2999, 2025-02-27:
I've got SH10RS+SBR192 which for several months I've used successfully with this library  (version 2024-11-02)

Main role is to handle forced charges based on conditions & times.

Last night automation ran with no errors but no forced charge occurred I manually set mode to forced charge this morning (see below) but again no effect on battery (though inverter mode in iCloud is changing to "Forced Mode" per screen snip below.

I'm not aware of anything changing in my setup & have no errors to work with.  HA Green being used, logs are clear, tried a restart.

Posting this here to see if anyone else having similar issues, maybe a change on Sungrow side ?

[image]

[image]

[image]

math2999, 2025-02-27:
I think I;'ve found issue - Set forced charge discharge power in W was somehow set to 200w (even though it's charging not discharging, that seems to affect the forced charge rate).

closed 2025-02-27 by math2999

Gnarfoz, 2025-02-27:
The name "Forced charge discharge power" is supposed to mean that this value is used for both charging and discharging, depending on "forced charge discharge cmd".

math2999, 2025-02-27:
> The name "Forced charge discharge power" is supposed to mean that this value is used for both charging and discharging, depending on "forced charge discharge cmd".

Yes thanks, I see that now. A bit of an ambiguous name though given there's also the "Set max battery charge power in W" "Set max battery discharge power in W" values. I wonder how they differ (as they were all set to 9KW) . Yaml code translating registers didn't yield me an answer.

Gnarfoz, 2025-02-27:
They are relevant for regular, non-forced mode operation.

The names are pretty much straight from Sungrow's Modbus documentation: https://github.com/Gnarfoz/Sungrow-Inverter/blob/main/Modbus%20Information/TI_20240924_Communication%20Protocol%20of%20Residential%20Hybrid%20Inverter-V1.1.5.pdf

math2999, 2025-02-27:
> They are relevant for regular, non-forced mode operation.
> 
> The names are pretty much straight from Sungrow's Modbus documentation: https://github.com/Gnarfoz/Sungrow-Inverter/blob/main/Modbus%20Information/TI_20240924_Communication%20Protocol%20of%20Residential%20Hybrid%20Inverter-V1.1.5.pdf

Thanks for taking the time to explain, much appreciated.