mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #488: some holding registers not working

GobblerGuy, 2025-03-06:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

Some of the registers (holding registers) have the value 65535 or a variant of this number. This seems an error to me. I cannot update the number in these holding registers by moving a slider. I get errors in the log,.

**Your Sungrow inverter:**
 - Model: [e.g. SH5K-30]
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
Core 2025.2.5
Supervisor 2025.03.0
Operating System 14.2
Frontend 20250221.0

**modbus_sungrow.yaml:**
 - Version/ time stamp : last update: 2025-02-15
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Load MKaiser integration
2. in my case I had to adjust some parameters to see values other than 'unavailable' for many registers.
    i've had the most success with:
    delay: 5 
    timeout: 20 
    message_wait_milliseconds: 100 
4. After waiting for the required period for all modbus registers to update, you can see that many holding registers have a value of 65535 or a decimal variant of. I cannot use the slider to update these registers.
5. See error
I do get some errors in the log file. See photo

I have tried loading a completely fresh installation of home assistant on another device. Despite connection issues that required a modbus proxy on my original system i was able to connect without modbus proxy in this fresh installation. in my orignal installation I was not able to get any holding registers at all to work. It was only in the fresh system that I experimented with the delay, timeout and message wait and found values that would allow some holding registers to work (such as minimum SOC). I pasted these in my old system and they worked there as well.

**Screenshots**

[image]

[image]

[image]

[image]

**Additional context**

GobblerGuy, 2025-03-08:
I have been looking at this a bit more today using QModMaster, and querying the input and holding registers above that give 65535 and also the pymodbus errors noted above, I also get 65535.  

for example:

[image]

So this result is completely separate to my Home Assistant setup and this repo. So is something to do with how my inverter is setup and not actually a bug?

Gnarfoz, 2025-03-08:
Looks like your inverter just doesn't support those registers, maybe. Did these work in the past?

closed 2025-03-13 by GobblerGuy