mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #498: Battery charging start power (W) slider grayed out since latest update

bsc76, 2025-03-17:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

Battery charging start power (W) slider grayed out since latest update

**Your Sungrow inverter:**
 - Model: SH20T
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)
   - [x] Waveshare

- Are you using a Modbus Proxy (mark one)
  - [x] yes
  - [] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2025.3.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2025-03-16 (see header of the file)
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to EMS control page
2. Check battery setting
3. See error slider grayed out

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

Gnarfoz, 2025-03-19:
That normally means the entity is unavailable.
Can you find `input_number.set_sg_battery_charging_start_power` in Settings > Devices > Entities?

bsc76, 2025-03-20:
Thanks for the hint @Gnarfoz - the entity exists but that question led me to view where I had the error again. When I checked the „bad“ entry I realised that the entity name was cut of at the end. Due to whatever reason…

Now it is working again. (Well, beside the fact that it is still questionable if the SH20T processes that entries for charging/discharging start at all.