mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #510: Export limitation returns an error

madisorg, 2025-04-07:
Hi,

running the scrips and all works like a charm. However i dont seem to get export limitation to work (which is my main objetive with this script actually to limit export to 0 at low prices).

The error i get is: Pymodbus: SungrowSHx: Error: device: 1 address: 13073 -> pymodbus returned isError True

I have the SH8 version and is on automatic update. 

I try update the value through local access to the inverter and works fine. My only nuance in the setup is that i have another Fronius inverter (15kW) connected to the same system (but cannot be set up as a slave for the Sungrow)...so my Sungrow ejected load values are all off but with the smart meter connection export limitation should still work (in my case i would set it to 23kW under normal circumstances and if price falls below the margin price then i want to set the limit to 0).

Thanks for any guidance.

Gnarfoz, 2025-04-07:
Hm, unable to reproduce that here, at least for now.
Just sanity-checking:
Are you maybe on some very old firmware?
Is the export limit actually enabled? (Export power limit mode)
Do you have access to some other Modbus client like mbpoll?
What does `mbpoll 192.168.0.61 -1 -r 13074 -t 4` report, and does `mbpoll 192.168.0.61 -1 -r 13074 -t 4 0` work?
Also, are you connected directly to the inverter's LAN port? (see https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/README.md#1-overview)

madisorg, 2025-04-09:
Just double-checked - sungrow firmware is up to date.
Through local access i can change the export limit value and this is successfully reflected in HA. However, controlling the value from HA returns the aforementioned error.
Also, i am connected to Sungrow though the LAN port.
I could not set up mpoll unfortunately as i could not make the connections (quite new to this). however HA modbus is working as i can read and write all other values that i have tried (eg setting min battery SOC, etc)

Gnarfoz, 2025-04-09:
Maybe a different client like QModMaster would work? It should not be more complicated than configuring this project. Entering the inverter's IP address & the relevant register number (13074) should be enough.

Otherwise I'm fresh out of ideas. 🤷‍♂
By "local access" you mean via the WiNet-S web interface? And then... which page exactly?
The "general parameter" page where you can read / write arbitrary Modbus registers?

heigo2, 2025-04-13:
@madisorg - did you manage to figure it out and get working? Seems i'm having similar issue with SH10RT. Worked before winder, but now has stopped and can't find root cause - installation is same as in autumn, no changes, i can set new limit values, sensor shows it is applied, but seems it is not written to register. 
integration as such works - when i change the "export power limit mode" value between enabled/ disabled, i see this value in registry changes (using ModbusTCP2MQTT integration to check values)

has there been maybe some change with registers? currently in modbus_sungrow.yaml it is set to 13073, changed it to 13074 and tried also with 13078, but same result. limit not applied.
Got now out of ideas what else to try

Gnarfoz, 2025-04-13:
Registers would only change with a new firmware version, and I don't think there's anything newer than 95.03 out for SHxRT?

madisorg, 2025-04-13:
An update:

- I also had a problem with SG not charging my BYD battery. Through support I found out my SG inverter was recalled due to some hardware issue. Do I get a newer refurbished version and also newer communication device (can't remember what the upgrade was exactly)
- I still have the old version and today I just tried to change the export limit and it somehow worked....only in the range from 0-7800W (I have 8kw inverter). No idea what had changed since a few days ago when it did not work.

Will wait for the new inverter to then keep on analyzing what's going on...

madisorg, 2025-04-13:
> [@madisorg](https://github.com/madisorg) - did you manage to figure it out and get working? Seems i'm having similar issue with SH10RT. Worked before winder, but now has stopped and can't find root [...]

Do you get error logs like I did? If not then we probably have two different problems...

heigo2, 2025-04-14:
Yep, seems something different - will keep looking - or if anybody could give some hints? :)
getting this: _"Error while executing automation automation.sungrow_inverter_export_power_limit_update: expected float for dictionary value @ data['value']"_

ABredhauer, 2025-04-20:
I have export limiting disabled but happened to try this today on my SH10RS and also get an error except mine is 134/0. Haven’t done any debugging yet but will look at it over the next few days as I have time

Velines, 2025-11-29:
I also stumbled upon not being able to set export limit, and I think it was fixed recently with 772a1d494b6f9568cd3debf7b18194535d122369. As @fgeorgsson pointed out the min export limit is the 3rd party installed PV capacity, and max is the sum. Both values can be read via Modbus from registers 5622 and 5623.