mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #526: Werte in Cloud und Integration unterscheiden sich

bennyrbg, 2025-05-08:
Hallo zusammen,

Ich habe einen SH10.0RT 112
Seit etwa 2-3 Wochen fällt mir auf, dass die Werte aus der Integration von Verbrauch und Erzeugung/Entladung um etwa 25% abweichen. Besonders Nachts fällt das auf, da nur die Batterie das Haus alleine versorgt.

[image]

In der Cloud ist der Wert korrekt und stimmt überein. Allerdings wird die Wirkleistung des WR ebenfalls niedriger angegeben. Laut meiner Recherche im PV-Forum hat Sungrow die Berechnung geändert, die Wirkleistung ist nunmehr die reine Netto-Leistung während der angezeigte (höhere) Wert der tatsächliche Verbrauch des Hauses inkl. Betriebsstrom und Wandlungsverlusten ist.

[image]

Dadurch ist natürlich die Darstellung fehlerhaft - wie kann ich das anpassen bzw muss das in der Integration angepasst werden?

HA Version war 2025.4.4 und jetzt 2025.5.0 auf OS15.2

Gruß

bennyrbg, 2025-05-08:
Der Verbrauch des Hauses war immer gleich der Wirkleistung.
Seit einem Update der FW durch Sungrow hat sich dies nun geändert.

Soll heißen, sobald der Wechselrichter etwas macht (er macht ja nur nichts wenn reiner Netzbezug und keine PV-Ladung vorliegt), wird nun der Eigenverbrauch sowie die Wandlungsverluste des Wechselrichters "ignoriert". Im obigen Beispiel sind das 100 Watt pro Stunde, die mir in der Statistik nun fehlen.

Ich suche daher nach einer Möglichkeit den tatsächlichen Hausverbrauch, ehemals sensor.load_power wieder korrekt darzustellen. Im Screenshot oben muss der nach meiner Logik wie in der iSolarcloud Batterieentladung = Aktueller Verbrauch sein.

Gnarfoz, 2025-05-09:
Welche Firmware-Version hast du?

Da es per Modbus keinen "Hausverbrauch" getrennt von "Last" gibt, kann man nur raten, woher iSolarCloud sich den Wert nimmt.
Ich habe mir einen Template-Sensor gemacht, der den Eigenverbrauch des WRs rudimentär ermittelt, indem ich von der PV-Erzeugung die Batterieleistung, die Last und den Export abziehe.
Das sind bei 10 kW PV, ~8,5 kW AC Last 1200 W Batterieladung immerhin ~360 W.

Alex84S, 2025-05-31:
Habe dieselbe Abweichung seit dem Update. Wie hast du es für dich gelöst?

andi-blafasl, 2025-08-06:
Mir ist diese Abweichung seit dem letzten Firmwareupdate auch gerade aufgefallen. Schön das ich nicht der einzige bin, der hier eine Abweichen iSolarCloud zu HomeAssistant sieht. Interessanterweise, liefert EVCC, das die Daten über den Winet-Dongle vom WR abruft, nachwievor die gleichen Daten die auch in iSolarCloud landen. Anscheinend hat sich das nur am Modbus-Register des WR geändert und der Winet liest was anderes aus.

[image]
Hier sieht man schön den Unterschied zwischen WR-Modbus (load power) und dem Winet via EVCC (Leistung Heim) wobei der Wert etwas gröber aufgelöst ist.

Was jetzt für die Statistik der bessere / ehrlichere Wert ist, weiß ich spontan auch nicht. Wäre interessant wie die Werte nachts ohne PV und bei leerer Batterie aussehe. Dann ist der Standby-Verbrauch des WR ja wirklich Last und Netzbezug den ich bezahlen muss 😉 

[image]
So sieht es aktuell nachts aus. Wobei ich da einen Unterschied von ca. 250W schon krass finde 🫨 und ich auch nicht Glaube, das mein Haus dann wirklich 0W verbraucht. Standby diverser Geräte und einiges an SmartHome und IT im Keller läuft ja immer 😉

closed 2025-12-06 by github-actions[bot]