mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #541: Typos in `modbus_sungrow_second_inverter_experimental.yaml`: from `hub`

greimel, 2025-06-19:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/98f8e65d565a4d39a2e579ed1ecafdd800cf4525/modbus_sungrow_second_inverter_experimental.yaml#L2707

I suppose this should read 

```
hub: SungrowSHx 2
```

here and in multiple other places. Since it seems the yaml is machine generated, I didn't bother submitting a PR. I hope that's ok.

Without this fix, I cannot adjust settings from homeassistant; with this fix I can.

greimel, 2025-06-19:
I hadn't seen that issue, sorry.

I am actually using `slave_id = 2`. 

My setup is as follows: The battery is connected to the master inverter, so I mostly interact with that one.

I am only interested in the second inverter, because I would like to implement zero-feed-in when prices are negative (I have a dynamic feed-in contract). I haven't been successful yet.

Gnarfoz, 2025-06-19:
I _think_ this makes sense: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/8193a2b39bfb2b65937e8a8b080ff88c68903efc

Gnarfoz, 2025-06-19:
Now, all the automations should use the Modbus hub defined at the top of the second inverter file.

greimel, 2025-06-19:
I assume I am using master/slave mode. Only inverter 1 talks to the smartmeter.

Your commits reflect my local changes exactly.