mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #543: Possible YAML issue? sg_battery_export_power_raw_3 and  export_power_raw_3

bleughb, 2025-06-21:
Hi,

Just playing around with the setup, trying to get the default dash items to work...
I think a register name is wrong,

Line 1186
     # this value returns a positive value when exporting and a negative value when importing power
      - name: Export power raw 3
        **unique_id: sg__battery__export_power_raw_3**

is only mentioned itself once to set up EXPORT POWER RAW 3

Should it be called
        unique_id: sg_export_power_raw_3

which is mentioned a bunch of other times, but doesn't seem to be setup anywhere?

Gnarfoz, 2025-06-21:
Hi, first of all I think you have multiple copies of the file in use. The _3 suffix indicates that. None of the sensors defined in modbus_sungrow.yaml end on _3.
Even line 1186 does not: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/99dbb1c6eb653c7072a366e7772278d260c01986/modbus_sungrow.yaml#L1186

Secondly, you have to differentiate between name, entity id, and unique id.
Sensor are never referred to by their unique id. They are always referred to by their entity id.
Sadly, there is no way to explicitly pre-define the entity id in YAML, it is automatically derived from the name, by replacing spaces with underscores etc.
The unique id is a purely internal thing and could be a random string for all intents and purposes. It's only there to allow customization of entities through the UI.

bleughb, 2025-06-21:
> Hi, first of all I think you have multiple copies of the file in use. The _3 suffix indicates that. None of the sensors defined in modbus_sungrow.yaml end on _3. Even line 1186 does not:

Thanks, I'd edited my questino - I have 3 inverters , so edited the YAML to have _3 on every ID
just having some odd issues getting some registers working!

Gnarfoz, 2025-06-21:
Well, like I said; you need to adjust the names, which will in turn result in changed entity IDs as well.
```yaml
name: Export Power Raw
```
should become
```yaml
name: Export Power Raw 3
```

Which would then be referred to in things like automations and scripts as 
```yaml
sensor.export_power_raw_3
```