mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #548: Invalid value 655350.0 returned for battery charging/discharging power – causes automation to fail

manuelwelker, 2025-06-28:
I'm currently experiencing a couple of issues when using the Sungrow integration with Home Assistant:

### Errors from the log:

```text
2025-06-27 18:59:16.427 ERROR (MainThread) [homeassistant.components.modbus.modbus] 
Pymodbus: SungrowSHx: Error: device: 1 address: 6226 -> pymodbus returned isError True

2025-06-26 12:29:28.233 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_charging_start_power_input_slider_update] 
Error executing script. Invalid data for call_service at pos 1: 
Invalid value for input_number.set_sg_battery_charging_start_power: 655350.0 (range 0.0 - 1000.0)

2025-06-26 12:29:28.558 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_discharging_start_power_input_slider_update] 
Error executing script. Invalid data for call_service at pos 1: 
Invalid value for input_number.set_sg_battery_discharging_start_power: 655350.0 (range 0.0 - 1000.0)

My System Setup:

Inverter: Sungrow SH20T V11 (3-phase hybrid inverter with backup/UPS functionality)

Battery: Sungrow SBH 100 (10 kWh high-voltage LiFePO₄ battery with backup capability)

Smartmeter: Sungrow Smartmeter V11

bennyrbg, 2025-06-30:
Got the same issue, my battery shows charging at around 65kW since around 2-3 days and I dont know how to fix.

bennyrbg, 2025-06-30:
It was updated on 2023 and i set it up in March 2024, so it was probably this version.

Gnarfoz, 2025-06-30:
> Inverter: Sungrow SH20T V11 (3-phase hybrid inverter with backup/UPS functionality)

Sounds like you have the SHxT firmware version where setting the battery (dis)charging power limit is not available. You could look into updating the inverter's firmware.

justusstiernacken, 2025-07-14:
You have to change the datatyp of "sg_battery_power_raw" from "uint16" to "int16".

Gnarfoz, 2025-07-14:
That wouldn't result in 0xFFFF being returned, would it?

manuelwelker, 2025-07-29:
@justusstiernacken sg_battery_power_raw ist already int16

closed 2025-11-28 by github-actions[bot]