mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #554: Option to ramp down PV production

s2719632, 2025-07-11:
**Is your feature request related to a problem? Please describe.**
I am looking to ramp PV production down to basically zero when the cost to buy power becomes negative. This would allow the opportunity to "buy" power from the grid for a small profit. My inverter today was export limited but still produced enough PV to meet the house load. My other inverter (fronius) was able to be ramped down to basically 0W PV production.

**Describe the solution you'd like**
A switch/slider for maximum PV production so that we could ramp that down to whatever we like. Alternatively, the option to completely turn off PV generation.

**Additional context**
SH10RT inverter
Amber electric Australia

Thank you for any insights into whether this is possible or not :)

Gnarfoz, 2025-07-11:
Well, the battery load speed is not relevant here, and so far the project does not include the ability to adjust the inverter's active power limitation. That's the feature request.
Luckily, this just recently added to Sungrow's Modbus register documentation, so it should be easy to do. Holding registers 13089 (active power limit mode (on/off)) and 13090 (active power limit ratio (%)) are the relevant registers.

Alternatively, you could just shut down the inverter. The toggle for that is already included.

joshbohmer, 2025-08-07:
@s2719632 lets chat on Discord to work this out and report back. This is a feature I've been chasing for a while! excited to see its been added in the most recent doc.

joshbohmer, 2025-08-07:
@ZEi20T we want no PV production at all... All house loads & battery charge should come from the grid even when solar is available.

Export Limit will still charge & power loads

joshbohmer, 2025-08-07:
@s2719632 reading the doc i think the Active Power Limit function will prevent AC output so house loads can be grid powered... but I suspect it won't limit PV->Battery DC flow. Still a step in the right direction to maximize negative buy prices.

joshbohmer, 2025-08-07:
13089 and 13090 return 0xFFFF for me also on both my SH15T. Bummer. Hopefully they enable it in a future FW, I'll be checking regularly.

joshbohmer, 2025-08-07:
Another that goes with it... I was playing with setting Active Power to 0% and my inverter shut down... WTF

That's another setting

31213 = Shutdown at active power limit ratio of 0%
0x00 AA [170] = On
0x00 55 [85] = Off

joshbohmer, 2025-08-09:
my code is so different it's basically just a new file, not ready for release yet. Here's the code block you'll need to get this feature working however, just adapt it to the current yaml layout.

```
      - name: Active power limitation raw
        unique_id: sg_active_power_limitation_raw
        device_address: *sg_device_address
        address: &reg_active_power_limitation_raw 31203 # reg 31204
        input_type: holding
        data_type: uint16
        scan_interval: 10
        state_class: measurement

      - name: Active power limitation ratio raw
        unique_id: sg_active_power_limitation_ratio_raw
        device_address: *sg_device_address
        address: &reg_active_power_limitation_ratio_raw 31204 # reg 31205
        input_type: holding
        data_type: uint16
        scan_interval: 10
        state_class: measurement
        scale: 0.1
        unit_of_measurement: "%"
        device_class: power_factor

      - name: APL shutdown at zero raw
        unique_id: sg_apl_shutdown_on_zero_raw
        device_address: *sg_device_address
        address: &reg_apl_shutdown_on_zero_raw 31212 # reg 31213
        input_type: holding
        data_type: uint16
        state_class: measurement
        scan_interval: 10

template:
  - sensor:
      - name: Active Power Limitation
        unique_id: sg_active_power_limitation
        device_class: enum
        availability: >
          {{ states('sensor.active_power_limitation_raw') not in ['unknown','unavailable'] }}
        state: >-
          {% set code = states('sensor.active_power_limitation_raw')|int(0) %}
          {% set map = { 
            0x00AA:'Enabled',
            0x0055:'Disabled'
          } %}
          {{ map.get(code, 'Unknown - should not see me! code: ' ~ code) }}

      - name: APL shutdown at 0%
        unique_id: sg_apl_shutdown_at_zero
        device_class: enum
        availability: >
          {{ states('sensor.apl_shutdown_at_zero_raw') not in ['unknown','unavailable'] }}
        state: >-
          {% set code = states('sensor.apl_shutdown_at_zero_raw')|int(0) %}
          {% set map = { 
            0x00AA:'On',
            0x0055:'Off'
            } %}
          {{ map.get(code, 'Unknown - should not see me! code: ' ~ code) }}

  - number:
      - name: Active Power Limit (W)
        unique_id: sg_active_power_limit_w
        min: "0"
        max: "{{ states('sensor.inverter_rated_output') | int(15000) }}"
        step: "100"
        availability: >
          {{ states('sensor.active_power_limitation_ratio_raw') not in ['unknown','unavailable']
             and states('sensor.inverter_rated_output') not in ['unknown','unavailable'] }}
        # Modbus → UI
        state: >-
          {% set base = states('sensor.inverter_rated_output') | float(15000) %}
          {% set pct  = states('sensor.active_power_limitation_ratio_raw') | float(0) %}
          {{ (base * pct / 100) | round(0) | int }}
        # UI → Modbus
        set_value:
          - variables:
              base: "{{ states('sensor.inverter_nominal_power') | float(15000) }}"
              pct: "{{ (value / base * 100) if base > 0 else 0 }}"
              pct_clamped: "{{ [ [pct, 100] | min, 0 ] | max }}"
              ratio_reg: "{{ ((pct_clamped | float(0)) * 10) | round(0) | int }}"
              enable_code: 0x00AA
              disable_code: 0x0055
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: *sg_device_address
              address: *reg_active_power_limitation_ratio_raw
              value: "{{ ratio_reg }}"
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: *sg_device_address
              address: *reg_active_power_limitation_raw
              value: "{{ enable_code if pct_clamped < 100 else disable_code }}"
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: *sg_device_address
              address: *reg_apl_shutdown_on_zero_raw
              value: "{{ disable_code }}"
          - action: homeassistant.update_entity
            target:
              entity_id:
                - sensor.active_power_limitation_ratio_raw
                - sensor.active_power_limitation_raw
                - sensor.apl_shutdown_at_zero_raw
```

Gnarfoz, 2025-12-03:
It is in the new version indeed, I've already created a pull request on https://github.com/bohdan-s/Sungrow-Inverter/pull/5 to add it there. :-)
Until then, it's also in my fork: [https://github.com/Gnarfoz/Sungrow-Inverter/raw/main/Modbus Information/TI_20251119_Communication Protocol of Residential Hybrid Inverter_V1.1.11_EN.pdf](https://raw.githubusercontent.com/Gnarfoz/Sungrow-Inverter/main/Modbus%20Information/TI_20251119_Communication%20Protocol%20of%20Residential%20Hybrid%20Inverter_V1.1.11_EN.pdf)

s2719632, 2025-12-03:
Thank you. Yes I am on SHxxRT so will have to wait I suppose

Rain4001, 2025-12-16:
> Ok... so I'm finally allowed to make this public, it's available in the most recent firmware and will end up in the next modbus spec: Disables ALL PV generation. Very useful in situations where grid prices go negative to allow forced charging from the grid (happens reasonably often in some areas of AU)
> 
> ```
>       - name: Cut PV generation inv1
>         unique_id: sg_cut_pv_generation_inv1
>         slave: *sg_device_address
>         address: &reg_cut_pv_gen 13017 # reg 13018
>         write_type: holding
>         command_on: 170
>         command_off: 85
>         scan_interval: 10
>         verify:
>           input_type: holding
>           address: *reg_cut_pv_gen
>           state_on: 170
>           state_off: 85
> ```
> 
> [@s2719632](https://github.com/s2719632) I think this is ultimately the answer to your original question

Couldn't get it to work on my SH10RS.
13017 - gave the error - Unexpected response from modbus device slave 1 register 13017, got 0xffff
31211 - no error but no change in state

s2719632, 2025-12-17:
No luck on either register on my RT model.

closed 2026-02-06 by github-actions[bot]

reubenb87, 2026-09-16:
Thanks @mkaiser and @theunknown86 !! I got 0 active power limitation working by adding these blocks into modbus_sungrow.yaml (release date 2026-06-19) after updating firmware of my inverter (SH20T with Sungrow Batteries)

Under template:number section:
```
      - name: Active Power Limit (W)
        unique_id: sg_active_power_limit_w
        min: "0"
        max: "{{ states('sensor.inverter_rated_output') | int(15000) }}"
        step: "100"
        availability: >
          {{ states('sensor.active_power_limitation_ratio_raw') not in ['unknown','unavailable']
             and states('sensor.inverter_rated_output') not in ['unknown','unavailable'] }}
        # Modbus → UI
        state: >-
          {% set base = states('sensor.inverter_rated_output') | float(15000) %}
          {% set pct  = states('sensor.active_power_limitation_ratio_raw') | float(0) %}
          {{ (base * pct / 100) | round(0) | int }}
        # UI → Modbus
        set_value:
          - variables:
              base: "{{ states('sensor.inverter_nominal_power') | float(15000) }}"
              pct: "{{ (value / base * 100) if base > 0 else 0 }}"
              pct_clamped: "{{ [ [pct, 100] | min, 0 ] | max }}"
              ratio_reg: "{{ ((pct_clamped | float(0)) * 10) | round(0) | int }}"
              enable_code: 0x00AA
              disable_code: 0x0055
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: !secret sungrow_modbus_device_address
              address: *reg_active_power_limitation_ratio_raw
              value: "{{ ratio_reg }}"
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: !secret sungrow_modbus_device_address
              address: *reg_active_power_limitation_raw
              value: "{{ enable_code if pct_clamped < 100 else disable_code }}"
          - action: modbus.write_register
            data:
              hub: *sg_hub_name
              slave: !secret sungrow_modbus_device_address
              address: *sg_reg_apl_shutdown_on_zero_raw
              value: "{{ disable_code }}"
          - action: homeassistant.update_entity
            target:
              entity_id:
                - sensor.active_power_limitation_ratio_raw
                - sensor.active_power_limitation_raw
                - sensor.apl_shutdown_at_zero_raw
```
and under template:sensor section:
```
      - name: Active Power Limitation
        unique_id: sg_active_power_limitation
        device_class: enum
        availability: >
          {{ states('sensor.active_power_limitation_raw') not in ['unknown','unavailable'] }}
        state: >-
          {% set code = states('sensor.active_power_limitation_raw')|int(0) %}
          {% set map = { 
            0x00AA:'Enabled',
            0x0055:'Disabled'
          } %}
          {{ map.get(code, 'Unknown - should not see me! code: ' ~ code) }}

      - name: APL shutdown at 0%
        unique_id: sg_apl_shutdown_at_zero
        device_class: enum
        availability: >
          {{ states('sensor.apl_shutdown_at_zero_raw') not in ['unknown','unavailable'] }}
        state: >-
          {% set code = states('sensor.apl_shutdown_at_zero_raw')|int(0) %}
          {% set map = { 
            0x00AA:'On',
            0x0055:'Off'
            } %}
          {{ map.get(code, 'Unknown - should not see me! code: ' ~ code) }}
```

To get it working if you have a battery, put the inverter into forced mode and set active power limitation to 0.

paulhomes, 2026-10-04:
> I suggest you try it still... I know an SH10RS which uses this feature. You can also try 31211 which is the register iSolarCloud uses for identical functionality.

Thanks @theunknown86 for the 31211 register info! I have an SH10RT on the latest firmware version and register 13018 returns 0xFFFF so I assumed it wasn't possible. However register 31211 returned 0x55 which sounded promising. I toggled register 31211 it between 0x55 (Allow PV) and 0xAA (Limit PV) and it did shutdown my panels. Toggling it back to 0x55 powered them back up again. I can now add an automation to do this when the Amber import price goes negative and save me going outside to turn them off manually!