mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #567: sensor.load_power shows 0 W at times although there is actual consumption

cablesky, 2025-08-19:
Describe the bug
The sensor sensor.load_power (house consumption) in Home Assistant sometimes reports 0 W, although there is actual consumption in the house. At the same time, the iSolarCloud app shows the correct house consumption.

To Reproduce
Steps to reproduce the behavior:

Connect Sungrow inverter directly via LAN cable (LAN port).

Configure the inverter’s LAN IP in Home Assistant (not the WiNet-S IP).

Observe sensor.load_power values in Home Assistant.

Occasionally the sensor reports 0 W, while iSolarCloud reports correct values.

Expected behavior
sensor.load_power should always show the correct house consumption (same as iSolarCloud).

Screenshots
(see attached screenshots)

Environment

Inverter model: SHT10RT-20
Connection: LAN (not WiNet-S)

Integration: mKaiser Sungrow SHx Modbus (LAN)

Additional context
It looks like the register might not always be read correctly, or there is a fallback to 0 happening.

[image]
[image]

cablesky, 2025-08-20:
Thanks for the reply.

Looking at the two screenshots I posted, it doesn’t seem to be caused by the 5-minute resolution of iSolarCloud. If you look closely, especially at image 1, you can see that iSolarCloud never shows a value of 0 (around 16:00).

Therefore, I also don’t think it’s an issue with the DTSU666 (which I do have in use).

I suspect it might be related to a firmware update of the inverter, as there is already another issue reported with a similar case.

It seems likely that the issue affects the direct reading of the Modbus registers via the LAN port and/or the WiNet-S port, while reading through iSolarCloud is probably not affected.

geisslerdominik, 2025-09-02:
I can agree, I do have the exact same behaviour.

[image]
[image]

geisslerdominik, 2025-09-02:
It might be only a feeling, but this only happens for low power values. If I switch on a high load (e.g. wallbox or a water boiler), the "0"-value dissappear and correct values are displayed.

mkaiser, 2025-09-02:
could you check the log files for potential modbus errors?

geisslerdominik, 2025-09-03:
> could you check the log files for potential modbus errors?

I have a brand new Screenshot attached.

[image]

Only modbus Errors i can see are as follows, but the timestamp does not match the "0"-value phase, and also the load-register (13007) is not included:

Logger: homeassistant.components.modbus.modbus
Quelle: components/modbus/modbus.py:300
Integration: Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 2. September 2025 um 20:14:48 (3821 Vorkommnisse)
Zuletzt protokolliert: 06:39:22

Pymodbus: SungrowSHx: Error: device: 1 address: 5740 -> Exception Response(132, 4, IllegalAddress)
Pymodbus: SungrowSHx: Error: device: 1 address: 13074 -> Exception Response(131, 3, SlaveFailure)
Pymodbus: SungrowSHx: Error: device: 1 address: 6226 -> Exception Response(132, 4, IllegalAddress)
Pymodbus: SungrowSHx: Error: device: 1 address: 33148 -> Exception Response(131, 3, SlaveFailure)
Pymodbus: SungrowSHx: Error: device: 1 address: 13086 -> Exception Response(131, 3, SlaveFailure)

geisslerdominik, 2025-09-03:
Maybe another input:
I use Modbus Proxy, as the sungrow inverter only deals with one client and my EVCC is the second client. So EVCC and the MKaiser Integration both point to the M

[image]

odbus proxi.
I use the LAN of the Winet-S2. 
SH15T.

What seems to be interesting: The EVCC sensor for Home Load does not show the "0"-Values

mkaiser, 2025-09-09:
Maybe it is also related to several race conditions in the underlying modbus Code. The modbus maintainer mentioned that there are several fixes on the way for the next releases. I guess in this case we should / can only wait