mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #575: Values still there after scan_interval

Bjoern3003, 2025-09-04:
Hi,

the script generally works well, and all values ​​arrive reliably in Home Assistant.

Unfortunately, when Home Assistant restarts or I reload YAML, the values ​​are always available only after the configured scan interval and are not initially loaded by Modbus. All values ​​are then unavailable.

Is this a known issue, and is there a solution?

Gnarfoz, 2025-09-04:
[The documentation for the Modbus integration](https://www.home-assistant.io/integrations/modbus/#configuring-multiple-connections) says:
> scan_interval integer (Optional, default: 15)
Update interval in seconds. scan_interval = 0 for no polling. **Entities are read shortly after startup and then according to scan_interval.** Remark, when restarting HA the last known value is restored.

So, no, in theory, they should be available at startup. However, this has also not been my experience.
You could create an automation that triggers on the HA startup event and calls `homeassistant.update_entity` on every sensor you care about.

Bjoern3003, 2025-09-13:
But it means next scan interval. I can reproduce it, by changing scan interval. Entitiesy with 10 minutes scan interval, only get her value after 10 minutes. Same with 1h interval.

Gnarfoz, 2025-09-13:
That's a bug in the Modbus integration, then. 🤷🏼‍♂️